misteraddons/Reflex-Adapt-Legacy #13: Add C64 / Amiga Mode

misteraddons, 2023-08-26:
Will this work using existing Genesis adapters? Is the pinout different from SMS?

rsn8887, 2023-08-30:
This would be awesome to have. Pinout:
https://allpinouts.org/pinouts/connectors/input_device/mouse-joystick-amiga-9-pin/

misteraddons, 2023-08-31:
Pretend I don't know much about C64 or Amiga. We would need to have new
adapters made and add a new mode for C64/Amiga. Do any controllers use all
3 buttons?

rsn8887, 2023-09-03:
No the maximum is two buttons.

Are you sure new adapters are needed? I thought DB 9 Pin should be compatible between Genesis and Amiga. See for example the Excel Sheet here: https://github.com/rsn8887/RetroAdapterMod/blob/master/misc/Retro%20Adapter%20connections.xlsx

rsn8887, 2023-09-03:
Here is some code:

https://github.com/rsn8887/RetroAdapterMod/blob/master/src_NormalAndArcadeMode/direct.c

misteraddons, 2023-09-04:
What's a good Amiga controller/joystick? There's a pin conflict between the
C64/Amiga controllers and Genesis. Check here to see that the VCC pin is
different:
http://wiki.icomp.de/wiki/DE-9_Joystick

rsn8887, 2023-09-06:
That is interesting, because I know that people used Mega Drive controllers on Amiga back in the day, and they worked.

The ArcadeR is a modern C64/Amiga controller. Atari 2600/VCS joysticks are also compatible. There is also the Competition Pro 9 Pin, but be careful, because USB version of the Competition Pro also exist.

I think the interface is super simple. It just uses separate microswitches to connect the respective wire to GND (when pressed) or leave it open circuit, when not pressed.

rsn8887, 2023-09-11:
Just to add another trivia: I can use my Competition Pro DB9 C64/Amiga controller with SNAX and the HDMI port to DB9 port cable on the Atari 7800 Core, I can play the Atari 2600/VCS version of Hero with it and it works just fine.

The same HDMI port to DB9 port cable also works with my original 6 button Genesis controllers using SNAX on the Genesis core.

misteraddons, 2023-09-13:
Do you have a Reflex Adapt? If so you can try building with the SMS mode
enabled in the Arduino IDE and flash your board that way :)

rsn8887, 2023-09-13:
> Do you have a Reflex Adapt? If so you can try building with the SMS mode enabled in the Arduino IDE and flash your board that way :)
> […](#)

Awesome, thanks for the hint. I didn't realize there was already this mode in the code, just not in the release.

rsn8887, 2023-09-15:
> Do you have a Reflex Adapt? If so you can try building with the SMS mode enabled in the Arduino IDE and flash your board that way :)
> […](#)

It works! I used Arduino IDE to flash my Reflex-Adapt with SMS Mode, and now it works with my Competition Pro Amiga 9DB Joysticks! Awesome. Maybe this could be added to the official release, since it already works like this using the Genesis cable?

misteraddons, 2023-09-16:
Great to know! I'll add some SMS firmware combos. Any other modes you care to include with the firmware combos?

rsn8887, 2023-09-16:
If possible, a combination NES/SNES + SMS + Genesis + PSX (+ NeGcon?) would be supercool but doesn’t have to be exactly that.

misteraddons, 2023-09-19:
Will give this a shot later tonight and close out the issue if it's
complete.

misteraddons, 2023-09-20:
Closed with https://github.com/misteraddons/Reflex-Adapt/commit/73593f91da43af64651eb776cd1a0a303c5ffafd

Added Combo 19

misteraddons, 2023-09-20:
Closed with commit https://github.com/misteraddons/Reflex-Adapt/commit/73593f91da43af64651eb776cd1a0a303c5ffafd

Added Combo 19

closed 2023-09-20 by misteraddons