misteraddons/Reflex-Adapt-Legacy #25: Fake Spinner sensitivity is inverted

sixtyfpsokthx, 2023-08-31:
I've found in playing Forgotten Worlds that setting the fake-spinner sensitivity lower (1) makes it incredibly touchy, and higher (8) allows for more breathing room.  This seems to be the opposite of the intent.

sixtyfpsokthx, 2023-08-31:
I should specify: **PSX-JogCon** Fake Spinner mode

sixtyfpsokthx, 2023-08-31:
Could it be, for weird math reasons, that I'm mistaken?  For example, the high (8) sensitivity often skips over some of the player-rotation values.

misteraddons, 2023-08-31:
Thanks for the feedback. In this case, 8 is the most sensitive. The number
has to do with the multiplier of pulse per revolution. The higher the
number, the more pulses per resolution. Think of it like mouse DPI
settings. A higher resolution mouse moves the cursor less far compared to a
lower resolution mouse moved the same distance. Here's what the manual says
about the JogCon sensitivity:

   1.

   Adjust spinner pulses per step (sensitivity) by pressing “mode” button
   while holding:
   1.

      Up = 1 (least sensitive)
      2.

      Right = 2
      3.

      Down = 4
      4.

      Left = 8 (most sensitive)

misteraddons, 2023-08-31:
Another note: the spinner sensitivity isn't tuned per game, it has the 4 available levels. The JogCon's spinner resolution isn't super high to begin with so the steps are definitely coarse.

sixtyfpsokthx, 2023-08-31:
Makes perfect sense.  If it could skip intervals, that would be warmly received.

closed 2023-08-31 by sixtyfpsokthx