misteraddons/Reflex-Adapt-Legacy #26: PSX SCPH-1200 Analog Not Working

Vorenu5, 2023-08-31:
The analog sticks on my SCPH-1200 PS1 controller are not registering correctly on PC or MiSTer. The analog LED on the controller is lit and the Reflex LCD displays dualshock mode for the controller.

I have tried modes from the following v1.05 firmwares:
Combo 12: PSX (My preferred combo)
Non-Nintendo: PSX+Gun
Combo 1: PSX+Gun+Jog

I've tested this same SCPH-1200 PS1 controller on a real PS1 and using a cheap PSX to USB adapter on PC and MiSTer and the analog sticks register as expected. I have also tested a SCPH-10010 PS2 controller and the analog sticks register correctly using the Reflex on PC. I'm currently not sure why the SCPH-1200 is not functioning correctly. Is this a possible firmware issue? I'm open to more testing if needed.

misteraddons, 2023-09-01:
I've seen one other report of this exact controller not behaving. I got one to test and mine works fine. Trying to think of what might be different about these.

Vorenu5, 2023-09-01:
I've tested the controller a bit further on PC and discovered if I hold one stick in a specific direction and move the opposite stick, some ranges of the stick will be detected.

For example: Holding right stick at the top-left diagonal will allow input ranges on the left stick to be detected when moved. Or if holding left stick at top-left diagonal, it allows input ranges on the right stick to be detected when moved.

mattd121, 2025-10-07:
Found my other grey 1200. Analogs don't work.