misteraddons/Reflex-Adapt-Legacy #27: Saturn Controller Issues

Vorenu5, 2023-08-31:
I've tested v1.05 Combo 12 Gen+Sat mode with 4 Saturn controllers with varying results:

1. MK-80100: Working as expected
2. MK-80116: Not detected by Reflex (Working on real Saturn console)
3. Retro-bit Wired: Buttons presses are detected, with phantom button presses on B3 & B6 which coincide with the entire button display on the Reflex flashing on and off (Working on real Saturn console)
4. Retro-bit Wireless: Similar to wired version except with longer phantom button presses on B3 & B6 with entire button display on the Reflex flashing at a higher rate (Working on real Saturn console)

I'm open to more testing if needed.

Vorenu5, 2023-08-31:
Update: After a bit of wiggling and seemingly getting the controller connector sitting at just the right angle, my MK-80116 is detected and working. Cleaning the controller connector contact points has not had an effect on the reliability of detection. I had a similar connector connection issue with some 3-button Genesis controllers as well but I won't bother creating a separate issue for something not seemingly software related.

misteraddons, 2023-09-01:
Yeah, Saturn connectors seem really sensitive. I've hand tested each Saturn adapter. I suspect there's variation from controller to controller that's hard to account for :(

Vorenu5, 2023-09-01:
I agree with the sensitivity of the connectors. Is there a possibility the behavior I'm experiencing with the Retro-bit controllers is a firmware issue?

daunator, 2024-01-29:
I can confirm the Retro-bit controllers issue. I am experiencing the same phantom button presses with the Retro-bit Wired controller.

misteraddons, 2024-02-09:
Sadly, Retro-Bit does weird things with the controller protocols. The Retro-bit controllers don't work with Daemonbyte either without inserting a massive delay.