misteraddons/Reflex-Adapt-Legacy #31: Wii PDP Wired Fight Pad Incorrect Analog Values

cmstar0, 2023-09-10:
Analog stick values on the PDP Wired Fight Pad for Wii read higher than when compared to the Raphnet v3 Wii Classic Controller Adapter.  

Analog down (on either stick) reads too high and the value appear to 'wrap' around to -127 (UP).  This makes the stick practically unusable when trying to press (DOWN).  

I've tested two separate controllers, the 'Yoshi' Fight Pad and the 'Mario' Fight Pad with the exact same results.  Value are also affected on the Wii Classic Controller Pro but not so far as to 'wrap' around to the wrong direction.

Attached are the screenshots comparing the two adapters.  Love this project, hope this helps for future improvements!

[image]

cmstar0, 2023-09-10:
@misteraddons If it helps or is needed, happy to send one of these controllers for testing.

misteraddons, 2023-09-11:
Do you know how to use the Arduino IDE? There's a setting in the code that sets the max GC Analog range. Anything over that range will be forced to +/-127. That shouldn't cause issues since digital +127 is a fixed value.

Either way, I'm happy to take a look at everything. Shoot me an email on my site's contact page. Thanks!