misteraddons/Reflex-Adapt-Legacy #40: Add support for N64 BlueRetro Adapter

Matt-Retrogamer, 2023-10-12:
Would it be possible to add support for the N64 Blueretro adapters that use Darkcloud's project ?
(https://github.com/darthcloud/BlueRetro)

Example of adapter: 

- https://8bitmods.com/n64-blueretro-bt-controller-receiver-with-memory-pak-original-grey/
- https://www.aliexpress.com/item/1005005008549068.html

I say N64 because that's the one I use with the N64 pad I just modded, but maybe this feature could maybe be extended to all Blueretro adapters.

Thanks!

misteraddons, 2023-10-12:
Hey, thanks for the request. Help me understand what you’re trying to
accomplish here. Any specific use case? Is it BT > N64 > USB? Just to make
pairing easier or for Switch compatibility?

Matt-Retrogamer, 2023-10-12:
Right, I should have started with that, it can sound weird 😁
The use case is MiSTer FPGA -> Reflex USB in N64 mode -> BlueRetro low latency Bluetooth adapter on the P1 -> Official N64 Switch Online Controller (or 8bitdo Bluetooth mod applied on an original N64 controller I'm my case).

It's a way to play wireless to the N64 Core and also have a save+rumble pack at all time (BlueRetro adapter for N64 include 4 virtual save packs).
The BlueRetro adapters keep all that at a very acceptable latency Bluetooth and it allows me to have only one wireless controller around to play both on original hardware and MiSTer.

Matt-Retrogamer, 2023-10-12:
(closed the issue by mistake, the use case is above 🤡)

misteraddons, 2023-10-12:
Have you verified that the BlueRetro N64 adapter doesn’t currently work?

Matt-Retrogamer, 2023-10-13:
Sure, that was the driver to open this case, I was surprised that it didn't work..

The status stays as "None" on both ports of the reflex USB when the BlueRetro is plugged-in (and paired to a controller). It does turn on and allows pairing + access to it's control web interface so I would say it's correctly powered.
If I plug a N64 wired controller instead, it's immediately detected and usable.

I've also done a cross check. If a plug the RetroBlue adapter I'm a N64, it works as expected (after initial pairing).

Vorenu5, 2023-10-13:
I've tested the BlueRetro N64 adapter you linked from 8bitmods, currently running firmware [v1.8.4 hw1 n64]. The Reflex Adapt detects it as a normal N64 controller for me (N64 ORIGINAL RANGE). I paired the official NSO controller and an original N64 controller with an 8bitdo modkit installed, and both work great on the latest N64 WIP MiSTer core. For the Reflex Adapt, I would not count on any save packs to work since those are emulating an actual controller pak, but they may work via SNAC in the future.

Vorenu5, 2023-10-13:
Additionally, I tested a DIY BlueRetro adapter that I built, currently running firmware [v1.8.3 hw1 universal]. With the Reflex Adapt in N64 mode, the BlueRetro firmware detects that its connected to an N64 and everything works with both controllers I previously tested with the adapter from 8bitmods (NSO & 8bitdo modkit).

Matt-Retrogamer, 2023-10-13:
Ok, interesting, for me it was never detected.
I'm on 1.8.3 hw1, I will try on latest with a factory reset and keep you posted 👍🏻

bryanparadis, 2024-10-04:
> Ok, interesting, for me it was never detected. I'm on 1.8.3 hw1, I will try on latest with a factory reset and keep you posted 👍🏻

Did it end up working?

bryanparadis, 2024-11-06:
> Nope, it never worked. I haven't tried with newer versions though, thanks for the reminder!

Did you get around to trying on latest Reflex Adapt version? @Vorenu5 reported it working so seems weird.