misteraddons/Reflex-Adapt-Legacy #42: SCP-1080 (PSX) controller issue with two controllers attached

cranium92838, 2024-01-03:
With two SCP-1080 controllers attached the UP button registers as UP+RIGHT and the down button registers as DOWN+LEFT.
This can be seen in the Reflex Adapt screen. It occurs when connected to MiSTer or PC.
With only one controller attached it works correctly.

cranium92838, 2024-12-14:
Hi,

It happens on both controllers and it still happens after I updated from 1.08 to 1.09.

I used Combo 12: NES+SNES, N64, Gen+Sat & PSX

Controller model is actually SCPH-1080.

 
 

Is it on both controllers that up is up+right and down+left? Does it still happen with current firmware?

cranium92838, 2024-12-14:
Also the same controllers both work perfectly with SNAX/SNAX64, so they're good.

bryanparadis, 2024-12-15:
What does name/mode does it display on the screen with one controller and with two controllers?

cranium92838, 2024-12-21:
Hi,

It displays "PSX" with one or two controllers connected.

I also have Dual Shock (SCPH-10010) controllers which work perfectly with two controllers connected.

closed 2026-02-21 by misteraddons