misteraddons/Reflex-Adapt-Legacy #50: 1.10 linux firmware updater - reflex-linux-x86_64 - doesn't detect reset button press and firmware flash cannot start

nobleEightfold, 2024-07-16:
With the 1.10 linux firmware updater ("reflex-linux-x86_64" in https://github.com/misteraddons/Reflex-Adapt/releases/download/v1.10/reflex-v1.10.zip) - while waiting at the "Press the RESET button on your Reflex device." prompt after choosing a firmware, the program does not respond when the reset button is pressed on the device, and the firmware flash is unable to start. I tried different combos, none of which made a difference in this behavior.

However, I was able to use the 1.08 version and the reset button press is detected and the firmware flash completes successfully.

misteraddons, 2024-07-16:
> With the 1.10 linux firmware updater ("reflex-linux-x86_64" in https://github.com/misteraddons/Reflex-Adapt/releases/download/v1.10/reflex-v1.10.zip) - while waiting at the "Press the RESET button [...]

Yep, I need to figure out what happened with the last release. I've been using 1.09 release myself.

The new output mode firmware is all but done. Need to work on a release for it once my testers give feedback and bugs are fixed :)

https://cdn.discordapp.com/attachments/1082669321555279952/1261718542617673829/reflex-mpg-v0.1_2.zip

bryanparadis, 2024-11-10:
```
Oct 14, 2023 Reflex-Adapt V1.09 - Last working updater
Dec 11, 2023 serialport-rs v4.3.0 - Released with a fix that breaks fallback device lookup: https://github.com/serialport/serialport-rs/pull/122
Several versions of Reflex-Adapt and serialport-rs
2024-10-21 serialport-rs v4.6.0 - Released with built in fallback for device enumeration without udev! https://github.com/serialport/serialport-rs/pull/220
Nov 4, 2024 Reflex-Adapt V2.01 - Still broken but the actual release assets are from like July/September not Nov 4! This threw me off
All builds that are actually since serialport-rs v4.6.0 seem to work fine on Linux I have not tested mister
```

So going forward Reflex-Adapt should be detected on reset on Linux again. Better support via serialport-rs v4.6.0 crate allows this to happen.

I think now though the fallback code [here](https://github.com/misteraddons/Reflex-Adapt/blob/main/reflex_updater/src/ports.rs#L62-L75) and [here](https://github.com/misteraddons/Reflex-Adapt/blob/main/reflex_updater/src/ports.rs#L21-L45) is  dead since serialport-rs v4.3.0.

- [ ] Confirm that reflex_updater built with serialport-rs v4.6.0+. Simply download [this](https://github.com/misteraddons/Reflex-Adapt/actions/runs/11693761837/artifacts/2149454958) and replace the file in v1.10.
- [ ] Up the version of serialport-rs to v4.6 in Cargo.toml to avoid older version. Possibly pin it to 4.6.X with `version = "~4.6"
- [ ] Remove dead code mentioned above if working on misterfpga or fix it if it's still necessary to keep

bryanparadis, 2025-02-05:
[armv7-unknown-linux-musleabihf(1).zip](https://github.com/user-attachments/files/18678131/armv7-unknown-linux-musleabihf.1.zip)

This should be a working linux updater for v2 firmware

bryanparadis, 2025-02-05:
One user in discord says no go on 1.10 and 2.0 firmwares on mister. Does not detect the reflex-adapt which lines up with my findings above. Maybe we should be doing a release and also taking down the one's that don't work.

bryanparadis, 2025-02-08:
On my new builds reflex.exe broke and it's because of serialport-rs 4.7.0! 4.6.1 works fine....