misteraddons/Reflex-Adapt-Legacy #61: Mapping GameCube Z button to BACK/Select button is problematic

InquisitiveCoder, 2024-11-14:
The GameCube's Z button is currently mapped to XInput's BACK button and Switch's minus button. This is an unfortunate choice since it doesn't match the default button layouts used by _Super Smash Bros. Ultimate_ or recently-released _Smash Bros. Melee_ clone, _Rivals of Aether 2_. What's worse, neither game allows binding actions to BACK/-, so the current GameCube mode renders the Z button _entirely_ unusable.

The current mapping doesn't seem to have much in the way of upsides either. As far as I know Steam's Big Picture Mode doesn't use the BACK button for anything and RetroArch only uses BACK as the equivalent of PageDown. Back/Select/- is far more likely to go unused in any given game than the bumpers as well. If someone uses GameCube controllers on MiSTer for whatever reason they can change the OSD hotkey to another button combination that doesn't involve Select.

My proposal is simple: map Z to RB (XInput)/R (Switch). This makes the GameCube mode work perfectly with _Smash Ultimate_ and - if the **Use Nintendo Button Layout** option is enabled on Steam - also _Rivals 2_. From poking around in `Input_GameCube.h` it even looks like this was already considered, since there's commented out code lines doing just that.

bryanparadis, 2024-11-22:
Trying to think what the reason for the current mapping is. No good history/blame hints unfortunately.

bryanparadis, 2024-11-22:
N64 is mapped the same way? https://github.com/misteraddons/Reflex-Adapt/blob/main/ReflexMPG/Input_N64.h#L342C1-L342C107

closed 2026-02-21 by misteraddons

misteraddons, 2026-02-21:
Updated prerelease firmware link (supersedes earlier prerelease links): https://github.com/misteraddons/Reflex-Adapt/releases/tag/v2.01-vb-nso-abfix-pre7

This pre7 build includes N64/PSX/Wii stability fixes plus prior SNES rumble/GameCube mapping/N64 timing updates. Please test and share results.