misteraddons/Reflex-Adapt-Legacy #76: Combo: Legacy 12 does not flash correctly

LordBarker80, 2025-06-24:
This was working perfectly before. Whenever I now flash combo 12 (legacy) SNES/NES VB, Gen/Saturn, N64 and PSX it doesn't seem to flash correctly and is limited to 'N64 original range' have tried re-flashing and resetting but no luck. All other combos seem fine. Thanks.

bryanparadis, 2025-07-08:
Can you describe what you mean by not flashing correctly?

LordBarker80, 2025-07-08:
I can do better than that: https://youtu.be/Py-_Tc4ZudM?si=MHT8LzHuyWvjWSSn&utm_source=MTQxZ

bryanparadis, 2025-07-10:
> I can do better than that: https://youtu.be/Py-_Tc4ZudM?si=MHT8LzHuyWvjWSSn&utm_source=MTQxZ

Thanks. The issue is pretty clear looking [here](https://github.com/misteraddons/Reflex-Adapt/blob/legacy_v1/Makefile#L66) as only N64 is actually setup for Combo 12!

bryanparadis, 2025-07-10:
[combo12 fix build.zip](https://github.com/user-attachments/files/21160531/combo12.fix.build.zip)

Here is a fixed build. I will need to open a PR to fix it for next release.

bryanparadis, 2025-07-11:
Should be all good now.

misteraddons, 2025-08-02:
Resolved in V2.01

closed 2025-08-02 by misteraddons