misteraddons/Reflex-Adapt-Legacy #79: Building Firmware Fails on MPGCombo10

luciusbono, 2025-08-03:
Trying to build the firmware on my Ubuntu 25.04 machine following the instructions on Github results in MPGCombo10 throwing an error saying it exceeds the maximum program space by 2 bytes. Is there a flag or something I'm missing?

```
arduino-cli -b Arduino-LUFA:avr:leonardo compile ReflexMPG --build-property "build.extra_flags={build.usb_flags} -DREFLEX_NO_DEFAULTS -DENABLE_REFLEX_SATURN -DENABLE_REFLEX_NEOGEO -DENABLE_REFLEX_N64 -DENABLE_N64_ANALOG_MAX=85 -DENABLE_REFLEX_GAMECUBE" -e --output-dir build/MPGCombo10
Sketch uses 28674 bytes (100%) of program storage space. Maximum is 28672 bytes.
Global variables use 1654 bytes (64%) of dynamic memory, leaving 906 bytes for local variables. Maximum is 2560 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.

Used library Version Path
EEPROM       2.0     /home/kingmetal/.arduino15/packages/Arduino-LUFA/hardware/avr/1.0/libraries/EEPROM
Arduino-LUFA 1.0.0   /home/kingmetal/.arduino15/packages/Arduino-LUFA/hardware/avr/1.0/libraries/Arduino-LUFA

Used platform    Version Path
Arduino-LUFA:avr 1.0     /home/kingmetal/.arduino15/packages/Arduino-LUFA/hardware/avr/1.0
Error during build: text section exceeds available space in board
make: *** [Makefile:32: firmware/MPGCombo10.hex] Error 1
```

luciusbono, 2025-08-03:
Update: realized I was building the latest commit to `main` and not the latest release. Switched to commit c04290bee9ec554a7c130e719e9af6fa69c43311 for `2.0.1` and now I'm getting passed MPG10 but failing on MPG13

```
arduino-cli -b Arduino-LUFA:avr:leonardo compile ReflexMPG --build-property "build.extra_flags={build.usb_flags} -DREFLEX_NO_DEFAULTS -DENABLE_REFLEX_SNES -DENABLE_REFLEX_WII -DENABLE_REFLEX_JAGUAR -DENABLE_REFLEX_PSX -DENABLE_PSX_GENERAL_OLED" -e --output-dir build/MPGCombo13
Sketch uses 31158 bytes (108%) of program storage space. Maximum is 28672 bytes.
Global variables use 2140 bytes (83%) of dynamic memory, leaving 420 bytes for local variables. Maximum is 2560 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.

Used library Version Path
EEPROM       2.0     /home/kingmetal/.arduino15/packages/Arduino-LUFA/hardware/avr/1.0/libraries/EEPROM
Arduino-LUFA 1.0.0   /home/kingmetal/.arduino15/packages/Arduino-LUFA/hardware/avr/1.0/libraries/Arduino-LUFA
SPI          1.0     /home/kingmetal/.arduino15/packages/Arduino-LUFA/hardware/avr/1.0/libraries/SPI

Used platform    Version Path
Arduino-LUFA:avr 1.0     /home/kingmetal/.arduino15/packages/Arduino-LUFA/hardware/avr/1.0
Error during build: text section exceeds available space in board
make: *** [Makefile:32: firmware/MPGCombo13.hex] Error 1
```

misteraddons, 2026-02-21:
Updated prerelease firmware link (supersedes earlier prerelease links): https://github.com/misteraddons/Reflex-Adapt/releases/tag/v2.01-vb-nso-abfix-pre7

This pre7 build includes N64/PSX/Wii stability fixes plus prior SNES rumble/GameCube mapping/N64 timing updates. Please test and share results.